robotology/poeticon #18: activeParticleTrack crashes when receiving two fixation points in rapid succession

gsaponaro, 2015-01-22:
Hi @vtikha, I'm using the activeParticleTracker module on this repository and I observed the following behaviour.

If my module sends two sets of coordinates in rapid succession to ```/activeParticleTrack/fixation:i```, the tracker crashes like this:
https://gist.github.com/gsaponaro/af0a9fade7e37939b35a

If I manually send the sets of coordinates via RPC, things go fine, like this:
https://gist.github.com/gsaponaro/fa0566bb745ef4302f16

At the moment, to circumvent this problem my module is using a delay of 1 second between writes to the fixation port.
https://github.com/robotology/poeticon/blob/master/src/worldStateManager/src/WorldStateMgrThread.cpp#L184

vtikha, 2015-01-22:
Hi @gsaponaro, ok thanks for reporting. I will dedicate the whole day tomorrow to finish and resolve some issues in the poetic modules. I will look at this asap and will also take the opportunity to fix the segmentations you mentioned a while ago.  
As soon as I have an update I will let you know.

gsaponaro, 2015-01-22:
OK, thank you. If anything is unclear, holler. :)

vtikha, 2015-01-27:
Hi @gsaponaro , I have committed some changes that will not allow this to happen. Please check it from your side and let me know. I have done my tests but I prefer having your confirmation before closing this issue

gsaponaro, 2015-01-29:
Confirmed working: now I can initialize the tracked objects super fast, thank you. :)

closed 2015-01-29 by gsaponaro