robotology/poeticon #22: activeParticleTrack: add option to number objects starting from a number other than 0

gsaponaro, 2015-02-06:
We are starting the OPC with the following entries

    item_0 (id 11) ((pos (0.0 0.0 0.0)) (is_hand true) (is_free true) (name lefthand))
    item_1 (id 12) ((pos (0.0 0.0 0.0)) (is_hand true) (is_free true) (name righthand))

and I need to "fuse" those IDs with the IDs coming from activeParticleTracker.

It would be helpful if activeParticleTracker had an option like ```--countFrom 13``` to assign track IDs starting from the number 13, instead of 0.

vtikha, 2015-02-07:
ok yes sounds good. Will do this during the trip tomorrow and we could test it on monday. Ciao!

vtikha, 2015-02-08:
@gsaponaro, I was looking at implementing this in a nice way and I thought of doing it in the following way:

 - add optional parameter to the track rpc command (or directly to the fixation port, in this case without the track command)
This means that instead of just sending "track fix_x fix_y" one can have the option of adding the "fromID" parameter as: "track fix_x fix_y fromID" if fromID is not set then it will start from 0 otherwise it will start from the number you set it to.

- otherwise go as you suggested with the countFrom command. 

Please let me know what suits you best.

gsaponaro, 2015-02-08:
I favour the ```--countFrom``` parameter option from command line, because I see it as a "global" setting of the tracker module/counter, not something track-specific or object-specific.

If you prefer to go for the rpc option, that will also work for me on the client side. I just would have to be careful to (a) specify fromID for the first object to track, don't specify that option for remaining objects; or (b) specify fromID for all objects, but make sure those numbers are in increasing order.

Safe travels and see you soon!

vtikha, 2015-02-08:
ok perfect. I have just committed the countFrom from the rpc interface. I have tested it quickly and behaves as you wanted. Let's test it tomorrow together some more.

gsaponaro, 2015-02-09:
Tested in bf27599530e787a072c0b5b3f92addca23caf311, it works fine, thanks.

closed 2015-02-09 by gsaponaro