robotology/poeticon #24: worldStateManager: check that the same ID is not used twice

gsaponaro, 2015-02-09:
Related to #22 

If OPC already contains id_50 and I ask the activeParticleTrack to ```countFrom 50```, worldStateManager crashes (it should print a warning instead).

Need to check for these cases in WSM, and possibly in activeParticleTrack too (re-using existing IDs).

vtikha, 2015-02-10:
I don't think that this should be changed in activeParticleTrack...There is actually no connection from APT to WSM but there is on the other side. Eg. APT does not know of the existence of WSM and therefore has no info on the IDs. And since WSB or somebody else sets " countFrom " it should make sure that no overlapping is happening .

gsaponaro, 2015-02-10:
That sounds right. I just stress-tested APT by tracking many objects and lowering ```countFrom``` several times between one track and another. No problems there.

Keeping open to remind me to test corner situations with IDs in the WSM<->OPC interaction.

closed 2015-03-09 by gsaponaro