robotology/poeticon #35: activeParticleTrack: cannot track nearby objects

gsaponaro, 2015-02-26:
Hi @vtikha, we are making some tests with these objects:

![image of six objects on table]
(http://i61.tinypic.com/2uzf1g7.png)

but we have trouble initializing activeParticleTrack to track all six of them (only two of them get initialized). worldStateManager is sending the fixations in rapid succession (see also #18). The fixation points are the coordinates of the centroids, obtained from segmentation:

    97.852310 103.968262
    52.069382 122.289307
    180.896347 88.432922
    158.374680 128.491791
    241.000000 127.000000
    217.500000 179.500000

I am trying to reproduce the problem with a datadump and will let you know the activeParticleTracker output.

The behaviour could be due to the crop widths and heights which are currently set to fixed values 30, 50 in function https://github.com/robotology/poeticon/blob/master/src/activeSegmentation/src/module/activeParticleTracker/src/activeParticle.cpp#L271

By the way, I noticed that activeParticleTrack allows two ways to track a new fixation point, which I guess are perfectly equivalent (please correct me if I'm wrong):
1. with the RPC command ```i32 track(1:i32 fix_x, 2:i32 fix_y)``` to rpc:i port, which returns the thread index
2. by sending ```fix_x fix_y``` to the fixation:i port

So far, worldStateManager was using method 2, but I think I will switch to method 1 which is more informative because of the return value. However, method 1 does not accept fractional coordinates which I have (see above), so either I round them to integers first, or we change the Thrift interface to accept floats... not a big issue, just pointing out a slight difference between the two methods here.

cc @lorejam

gsaponaro, 2015-02-26:
Update: the initialization problem was due to the too fast fixations (like in #18), not due to crop window sizes. Inserting a delay of 1 second between initialization solved this particular problem.

vtikha, 2015-02-27:
@gsaponaro Yes there is two ways of allowing to track a new fixation point but both ways call the same function. I have made the changes you requested for method 1 to accept fractional coordinates. 
Can you please tell me at which speed you are sending the fixation requests? This was fixed before last visit and you also tested it locally. One second is too much.

gsaponaro, 2015-02-27:
Thanks, how do I quickly measure the speed at which my module is streaming the requests? Is there a command-line yarp tool for that, or do I need to modify my code with yarp::os::Time and compute differences?

vtikha, 2015-02-27:
exactly just simply use yarp::os::Time and compute it.

gsaponaro, 2015-02-27:
Trying to initialize multiple fixations as fast as I can (no delay), I observe the following in my module. At the end of the process, only 2 out of 5 objects (the "big" blobs above a certain area threshold) get actually fixated:

![sending fixations fast]
(http://i60.tinypic.com/33vnlf5.png)

Corresponding output of activeParticleTrack:

https://gist.github.com/gsaponaro/58fc59da6264d634689b

In this test, my module is doing this and sending the fixation requests to fixation:i port (method 2):

    for(a : all blobs in segmentation)
    {
        u = inAff->get(a+1).asList()->get(0).asDouble();
        v = inAff->get(a+1).asList()->get(1).asDouble();
        trackerCmd.clear();
        trackerCmd.addDouble(u);
        trackerCmd.addDouble(v);
        yDebug() << __func__ <<  "sending fixation request to tracker:" << trackerCmd.toString().c_str();
        double start = Time::now();
        trackerPort.write(trackerCmd);
        yDebug("time elapsed: %f seconds", (Time::now() - start) );
    }
    yInfo("done initializing tracker");

vtikha, 2015-03-03:
@gsaponaro This definitely cannot work like that. You are sending a command on average 90 microseconds and expect it to work? This has nothing to do with an area threshold nor crop window sizes. A good procedure would be to send this to the rpc for synchronisation.If you still want to use the fixation:i port, then you should have a delay on the port (before the next write) to allow the callback to elaborate the message. 1 second is the extreme opposite so I suggest you in the range of 0.1 to 0.5 seconds and see what works best.

gsaponaro, 2015-03-03:
Thanks. Effectively, I don't need the fixation port anymore (method 2) and I switched to using the RPC port last week (method 1) which works like a charm without explicit delays on the client side because of the blocking nature of synchronous RPC.

Indeed, I had reached your same conclusion in my second message of this thread ("the initialization problem was due to the too fast fixations (like in #18), not due to crop window sizes"). Closing issue. :)

closed 2015-03-03 by gsaponaro