robotology/poeticon #36: activityInterface: icubmod linking problem

gsaponaro, 2015-03-04:
Using robotology/icub-main@661d2ae41273a6c1c8dc1ae87fd882cec5b8af28, **activityInterface** compiles fine.

However, using the changes introduced in robotology/icub-main@c5f86fdf36395c21ba9dce6b743763aa7985c035, compilation breaks here:

```
~/poeticon/build$ make
[ 10%] Building CXX object src/activityInterface/CMakeFiles/activityInterface.dir/src/memoryReporter.cpp.o
[ 20%] Building CXX object src/activityInterface/CMakeFiles/activityInterface.dir/src/activityInterface.cpp.o
[ 30%] Building CXX object src/activityInterface/CMakeFiles/activityInterface.dir/src/main.cpp.o
[ 40%] Building CXX object src/activityInterface/CMakeFiles/activityInterface.dir/src/activityInterface_IDLServer.cpp.o
Linking CXX executable ../../bin/activityInterface
/usr/bin/ld: cannot find -lcartesiancontrollerserver
/usr/bin/ld: cannot find -lcartesiancontrollerclient
/usr/bin/ld: cannot find -lgazecontrollerclient
collect2: error: ld returned 1 exit status
make[2]: *** [bin/activityInterface] Error 1
make[1]: *** [src/activityInterface/CMakeFiles/activityInterface.dir/all] Error 2
make: *** [all] Error 2
```

Verbose information:

```
Linking CXX executable ../../bin/activityInterface
cd /home/gsaponaro/poeticon/build/src/activityInterface && /usr/bin/cmake -E cmake_link_script CMakeFiles/activityInterface.dir/link.txt --verbose=1
/usr/bin/c++   -O3 -DNDEBUG    CMakeFiles/activityInterface.dir/src/memoryReporter.cpp.o CMakeFiles/activityInterface.dir/src/activityInterface.cpp.o CMakeFiles/activityInterface.dir/src/main.cpp.o CMakeFiles/activityInterface.dir/src/activityInterface_IDLServer.cpp.o  -o ../../bin/activityInterface -rdynamic /home/gsaponaro/yarp/build/lib/libYARP_OS.so /home/gsaponaro/yarp/build/lib/libYARP_sig.so /home/gsaponaro/yarp/build/lib/libYARP_math.so /home/gsaponaro/yarp/build/lib/libYARP_dev.so /home/gsaponaro/yarp/build/lib/libYARP_name.so /home/gsaponaro/yarp/build/lib/libYARP_init.so /home/gsaponaro/icub-main/build/lib/libicubmod.a /home/gsaponaro/yarp/build/lib/libYARP_sig.so /home/gsaponaro/yarp/build/lib/libYARP_OS.so -lcartesiancontrollerserver -lcartesiancontrollerclient -lgazecontrollerclient -Wl,-rpath,/home/gsaponaro/yarp/build/lib::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: 
/usr/bin/ld: cannot find -lcartesiancontrollerserver
/usr/bin/ld: cannot find -lcartesiancontrollerclient
/usr/bin/ld: cannot find -lgazecontrollerclient
collect2: error: ld returned 1 exit status
make[2]: *** [bin/activityInterface] Error 1
make[2]: Leaving directory `/home/gsaponaro/poeticon/build'
make[1]: *** [src/activityInterface/CMakeFiles/activityInterface.dir/all] Error 2
make[1]: Leaving directory `/home/gsaponaro/poeticon/build'
make: *** [all] Error 2
```

vtikha, 2015-03-04:
yes it seems that there is a problem with some recent changes introduced in cub-main.
I think they are working on it to resolve the issue. I will keep you posted

vtikha, 2015-03-05:
Please try the compilation again. pull & do a clean make please.

gsaponaro, 2015-03-05:
Just tried that for yarp, icub-main and poeticon (I made sure to delete the build dir completely). Same error.

vtikha, 2015-03-05:
Then I do not know. Did you change something on your machine? Nothing has been changed from this end.

gsaponaro, 2015-03-05:
Nope, I just tried with a clean new username on another machine.

If you compile the latest icub-main and poeticon builds, do you not get this error for activityInterface?

vtikha, 2015-03-05:
ok I was told it worked but obviously it does not yet. I will push people

vtikha, 2015-03-05:
please update icub-main. The error has been changed.

gsaponaro, 2015-03-05:
Thanks @vtikha and @drdanz, indeed robotology/icub-main@be1af160f1ed3b1ecab4a874b6c4db47f3e8bdb0 fixed it.

closed 2015-03-05 by gsaponaro