robotology/poeticon #39: blobSpotter ported in IOL

pattacini, 2015-03-11:
Hi guys,

I and @vtikha decided to port `blobSpotter` in [IOL](https://github.com/robotology/iol), the place where it is more natural to have it (i.e. alongside with `blobExtractor`). IOL is also a dependency for `poeticon`, somehow; so, there shouldn't be any problem.

Remember to remove the copy of `blobSpotter` in this repo then, whenever you have time :)

pattacini, 2015-03-22:
I've ported the recent change introducing the `topBound` option into [`iol/blobSpotter`](https://github.com/robotology/iol/tree/master/src/blobSpotter]).

@gsaponaro @vtikha please consider stopping working with this version of `blobSpotter`.

The new one has also the [documentation](http://robotology.github.io/iol/doxygen/doc/html/group__blobSpotter.html).

gsaponaro, 2015-03-23:
Thanks @pattacini for the reminder.

I confirm that the latest changes ([this fix](https://github.com/robotology/poeticon/commit/4ed5893bfe18a401613373953f21f513c876bbbd) and the topBound option) are ported.

@vtikha can I go ahead and remove the module from the poeticon repo?

vtikha, 2015-03-23:
@gsaponaro be my guest :-)

commit 619c38e6 (2015-03-23): blobSpotter moved to https://github.com/robotology/iol

Fixes #39
closed 2015-03-23 by gsaponaro