robotology/poeticon #147: activeParticleTrack not closing gracefully

gsaponaro, 2015-07-01:
Hi @vtikha, we noticed that **activeParticleTrack** does not exit properly on ctrl+c, at least when it is tracking more than 1 objects. For now we are just killing and restarting the module as a workaround.

Steps to reproduce:

* launch tracker, connect an image to the image:i port
* initialize a couple of (valid) fixations, for example
```
$ yarp rpc /activeParticleTrack/rpc:i 
>>track 100.0 100.0
Response: 0
>>track 120.0 120.0
Response: 1
```
* try to quit the module with ctrl+c, here is what happens:
```
starting the shutdown procedure
cleaning up...
attempting to interrupt ports
finished interrupt ports
now closing ports...
Going to stop 2 Threads
[TRACKER HANGS HERE]
^C[try 2 of 3] Trying to shut down
^C[try 3 of 3] Trying to shut down
^CAborting (calling abort())...
^CAborted (core dumped)
```

vtikha, 2015-07-07:
@gsaponaro I have just committed something that seems to have fixed the problem. I will still look at it as I am not entirely convinced :-). Please have a go at it and let me know if it works for you.

gsaponaro, 2015-07-08:
Works fine here, thanks. :)

closed 2015-07-08 by gsaponaro