robotology/poeticon #156: Add possibility to clear stack of objects

gsaponaro, 2015-09-17:
It would be useful if **activityInterface** had an RPC function to clear/empty the stack of objects. Currently, the stack is implemented as a ```std::map<string, cv::Scalar>``` that only supports adding items, not removing them.

gsaponaro, 2015-09-18:
Hi @vtikha , would it be possible for you to implement this? Or if you're very busy, give us a few pointers so we can change the code ourselves?

Use cases:

1. Human sabotage scenario. When the robot has almost completed the sandwich (bun-bottom and some ingredients are already stacked correctly) and has the bun-top in its hand about to achieve the goal, we remove the ingredients that are on the bun-bottom. At this point (for now manually via RPC) we would issue the ```clearStack``` function we are talking about;
2. To start a plan after a previous plan was completed, currently we have to restart **activityInterface** every time. With this function we would solve this issue.

/cc @AlexAntn @lorejam

vtikha, 2015-09-19:
@gsaponaro, yes sure I will implement this feature. I also think that it will even work if the sabotage scenario is not a complete removal of all ingredients but only one. The stack will be emptied, and the planner will re-plan accordingly with what is available. Seems ok :+1:

vtikha, 2015-09-22:
@gsaponaro added the requested feature. I have tested it offline (without a robot) and therefore waiting for your confirmation before closing this.

gsaponaro, 2015-09-22:
Thanks @vtikha, I tested 8f6f805e8cb546e5e19ff43aac179bc37ff3be20.

```resetObjStack``` does the reset correctly, however there are issues when re-making a new stack after having reset it, here are the sequences of actions that I tried:

* take cheese left, put cheese bun-bottom -> ```underOf cheese``` returns ```(bun-bottom)``` -> I reset the stack and now ```underOf cheese``` returns ```()```, OK
* now I re-do take cheese left, put cheese bun-bottom -> ```underOf cheese``` returns ```()```, instead of the expected result ```(bun-bottom)```

vtikha, 2015-09-22:
@gsaponaro I have committed and tested a minor change and here it all works fine.

gsaponaro, 2015-09-22:
Yep, in 4b01a32ff20a7c648d217eb91ae2a3c31ef64c8d the functionality is perfect, thanks!

closed 2015-09-22 by gsaponaro