robotology/poeticon #157: Commit praxInterface module (interface between PRAXICON and YARP)

gsaponaro, 2015-09-22:
Hi @vtikha, I suggest this module gets committed to the repository. It should probably be compiled only on WIN32 systems. Do you agree?

If you want, I can commit the version that we have on our Windows/PRAXICON laptop from the latest integration meeting.

vtikha, 2015-09-22:
@gsaponaro agreed

gsaponaro, 2015-09-22:
OK, I committed **praxiconInterface**.

Even though it will be used in Windows, I tried to compile it in Linux for testing. There were two problems, I solved the first one but not the second one.

First problem was:
```
/home/gsaponaro/NOBACKUP/poeticon_upstream/src/praxiconInterface/src/praxiconInterface.cpp: In member function ‘virtual bool PraxiconInterface::updateModule()’:
/home/gsaponaro/NOBACKUP/poeticon_upstream/src/praxiconInterface/src/praxiconInterface.cpp:78:19: error: no match for ‘operator!=’ (operand types are ‘yarp::os::Bottle’ and ‘long int’)
         if (speech!=NULL)
                   ^
```
which I solved by changing ```(speech!=NULL)``` to ```(!speech.isNull())```, where ```speech``` is a ```Bottle```.

Second compilation issue:
```
$ make
[ 25%] Building CXX object CMakeFiles/praxiconInterface.dir/src/stdsoap2.cpp.o
/home/gsaponaro/NOBACKUP/poeticon_upstream/src/praxiconInterface/src/stdsoap2.cpp:71:54: fatal error: include/iCub/PraxiconTreePortBinding.nsmap: No such file or directory
 #include "include/iCub/PraxiconTreePortBinding.nsmap"
```

I added ```include/iCub/PraxiconTreePortBinding.nsmap``` to ```CMakeLists.txt``` but it didn't help. Any clues? Can you try compiling?

vtikha, 2015-09-22:
This module wa intenses to be used only in windows due to its depend

gsaponaro, 2015-09-22:
Yep, this must run on the same (Windows) machine where the other PRAXICON modules run.

It was a stupid problem, I had put one of the files in the wrong location... fixed now, tested and it works. Thanks.

closed 2015-09-22 by gsaponaro