robotology/poeticon #158: Objects that are available in IOL are not available in PRAXICON / praxInterface

gsaponaro, 2015-09-22:
Hi @vtikha again :) I quote an issue from the 27-28 Jul email thread with Panagiotis:

*The recognised objects that are not used by the Praxicon reasoner sound like a problem in the praxiconInterface or the communication of the object recognition module rather than the reasoner itself as the printouts of the reasoner are only visible through the netbeans environment currently.
I believe Vadim can figure it out or perhaps you could add some custom printouts in the object recognition module or the praxiconinterface to see when are the objects lost.*

Probably we should first take care of #157, then this.

vtikha, 2015-09-22:
@gsaponaro this seems very odd as it does not depend on the praxIterface. If the objects (at the time of the communication with the praxicon) are not recognised properly then they will not be forwarded to the proxicon. With the new recognition module this is not likely to happen. In any case, do #157 and I will double check this.

gsaponaro, 2015-09-22:
OK, #157 now done.

The "lost/undetected objects" problem had occurred with the new IOL Caffe recognition module. More details in the email "Praxicon available objects".

/cc @AlexAntn

vtikha, 2015-09-23:
I confirm that the `praxiconInterface` is not responsible for the missing objects as its job is to get the query, formulate it in a praxicon readable way and communicate with it. It all depends on the recognition pipeline and the activityInterface. As things have changed there we need to make sure all is still consistent.

vtikha, 2015-09-28:
I have simplified the activityInterface and did some tests and all seem good. I am not missing any objects in repeated experiments. Please pull and let me know.

gsaponaro, 2015-09-28:
Thanks, it works now!

Just for reference, I tested with the following setup:

[image]

[image]

Before today, we used to observe the following:

```
>>askPraxicon "make a sandwich"
...
[INFO][askPraxicon] tool names: rake rake
[INFO][askPraxicon] object names: cheese stick rake bun-top bun-bottom meat tomato 
[INFO][askPraxicon] sending 
 (available cheese stick bun-top bun-bottom meat tomato) (query "make a sandwich") (missing) 
```
^ note that rake was not being propagated; also, the part "tool names: rake rake" looks wrong

From today (https://github.com/robotology/poeticon/commit/9aab7ec222e935b309de0496972c20fc91ad6ddd), we now get:

```
>>askPraxicon "make a sandwich"
...
[INFO][askPraxicon] request is : make a sandwich 
[INFO][askPraxicon] object names: cheese stick rake bun-top bun-bottom meat tomato 
[INFO][askPraxicon] sending 
 (available cheese stick rake bun-top bun-bottom meat tomato) (query "make a sandwich") (missing) 
```
with rake correctly propagated. Closing issue.

PS: I know you're super busy, and this is quite a small project, but if it's not too much effort, could you put the #ISSUE-NUMBER in your commit message texts? This creates an automatic reference in the page for that issue, it's quite helpful and clear to help tracking development history, as sometimes I am testing old vs new versions of programs to see if a problem is consistent, or when it originated. :)

vtikha, 2015-09-28:
@gsaponaro ok perfect, thanks for the update. And will also put the #ISSUE-NUMBER as suggested. Although this is a very small project it is still a good idea. :+1:

vtikha, 2015-09-28:
The missing objects was correct. Tools should be removed from the list before sending the query back to the praxicon, as tools should not be part of the sandwich making and the praxicon cannot deal with them.

gsaponaro, 2015-09-28:
OK for tools being removed from the list.

Not OK for/if common objects being removed from the list. But I don't remember if this case was also happening, do you @AlexAntn ? (I.e. one of the sandwich stack ingredients not being propagated). I guess we will find out with further tests.

vtikha, 2015-09-28:
@gsaponaro NOT ok for common objects and this is clear. I was looking at the function and that was what it was trying to do. The common objects are not removed in any way. The attempt was to remove only tools but with some bugs obviously as you also showed in the above problem (what used to happen): tool names were wrong `rake rake` and therefore `stick` was not considered as tool and thus propagated `(available cheese stick bun-top bun-bottom meat tomato)`

AlexAntn, 2015-09-28:
Hello,

There was a bug (around 4 months back) that the Praxicon would not
recognise some ingredients if there was no tool present, but I haven't seen
that since.

There is another bug, though, where the Praxicon uses the stick as an
ingredient of the sandwich, when it should not be considered.

Regards,
Alex

vtikha, 2015-09-28:
@AlexAntn This is what we were talking about in this issue and in #164.

vtikha, 2015-09-29:
@gsaponaro I have committed the changes we mentioned above and returned to the old situation. I have done many tests and everything was working as expected and there were no issues on the naming etc. I also tried your example above and as I mentioned at the beginning this is a recognition problem (this is clear) eg:

[image]

```
>>askPraxicon "make a sandwich"
...
[INFO][askPraxicon] tool names: rake rake
[INFO][askPraxicon] object names: cheese stick rake bun-top bun-bottom meat tomato 
[INFO][askPraxicon] sending 
 (available cheese stick bun-top bun-bottom meat tomato) (query "make a sandwich") (missing) 
```

Now, recognition (label) is perfect yes, the objects are correctly recognised, therefore why the reply is tool name `rake rake`? Since we do not know what the name of the tool is (only general obj labels), I attempt to identify them visually. Once I have identified the blobs that look like tools, I query the recognition module with a 2D point (cog of the blob). Then the nearest blob is located and the recognition is done on that one. Now, if the blobs are too close the cog of one blob might be closer to the other one then a wrong blob is selected for recognition hence the two tools names `rake rake`. Obviously `stick` is not considered a tool and not removed from the list. I will now try to see if there is an issue with the blobs and cog but I will suggest for the meantime to make sure the objects (especially the tools) are not close to each other.

gsaponaro, 2015-10-02:
Hi @vtikha, when you have time, here I report a more detailed test regarding this morning's problem about distinguishing tools from ingredients. As you can see, everything is well separated in terms of distances between 2D blob centres of gravity, etc.:

[image]

```
>>askPraxicon "make a sandwich"
Response: (((hand grasp cheese) (cheese reach bun-bottom) (hand put cheese) (hand grasp stick) (stick reach cheese) (hand put stick) (hand grasp bun-top) (bun-top reach cheese) (hand put bun-top)))

[INFO][askPraxicon] request is : make a sandwich 
[INFO]the dffs are 86 12 total 98
[INFO]the dffs are 235 3 total 238
[INFO]the dffs are 2 4 total 6
[INFO]the dffs are 158 24 total 182
[INFO]the dffs are 160 40 total 200
[INFO][askPraxicon] tool names: rake 
[INFO][askPraxicon] object names: cheese stick rake bun-top bun-bottom 
[INFO][askPraxicon] sending 
 (available cheese stick bun-top bun-bottom) (query "make a sandwich") (missing) 
```

Problems:
1. stick is not considered a tool
2. also, when the bun-top is far, we get:
```
>>reachableWith bun-top
Response: (rake) <-- stick not present
```

For comparison, when the bun-top is near, we get the expected result (according the current version of activityInterface):
```
>>reachableWith bun-top
Response: (cheese stick rake bun-bottom left right) <-- OK, stick is present
```

vtikha, 2015-10-16:
@gsaponaro I have tried this in many different configurations and always had a good outcome. 
here are some examples: (obviously the labels are totally useless for the actual sandwich making but for the tests of the tools its more than enough)

[image]

```
INFO][askPraxicon] tool names: Shuvel Rake 
[INFO][askPraxicon] object names: Shuvel Octopus Car Rake Ball 
[INFO][askPraxicon] sending 
(available Octopus Car Ball) (query "make a sandwich") (missing) 
```

[image]

```
INFO][askPraxicon] tool names: Shuvel Rake 
[INFO][askPraxicon] object names: Shuvel Octopus Car Rake Ball 
[INFO][askPraxicon] sending 
(available Octopus Car Ball) (query "make a sandwich") (missing) 
```

I have committed a lot of debugging info to make sure why this is not working correctly at your end. 
Please try it again when you'll have time and send me the terminal output and a object configuration image.

gsaponaro, 2015-10-16:
Thanks @vtikha, here is a fresh tests with the new debug prints.

[image]

```
>>askPraxicon "make a sandwich"
Response: (((hand grasp cheese) (cheese reach bun-bottom) (hand put cheese) (hand grasp stick) (stick reach cheese) (hand put stick) (hand grasp bun-top) (bun-top reach cheese) (hand put bun-top)))
```

```
[INFO][askPraxicon] request is : make a sandwich 
[DEBUG][getToolLikeNames] area 1815.000000 
[DEBUG][getToolLikeNames] area 1455.000000 
[DEBUG][getToolLikeNames] area 1136.000000 
[DEBUG][getToolLikeNames] area 821.000000 
[DEBUG][getToolLikeNames] area 1777.000000 
[DEBUG][getToolLikeNames] area 1169.000000 
[DEBUG][getToolLikeNames] area 2785.500000 
[DEBUG][getToolLikeNames] area 1965.500000 
[DEBUG][getToolLikeNames] area 612.500000 
[DEBUG][getToolLikeNames] area 404.500000 
[ERROR][getToolLikeNames] allLength size 7 
[INFO][getLabel] cog  160 128
[INFO][getLabel] pos  81 145
[INFO][getLabel] the dffs are 79 17 total 96
[INFO][getLabel] cog  235 140
[INFO][getLabel] pos  81 145
[INFO][getLabel] the dffs are 154 5 total 159
[INFO][getLabel] cog  84 150
[INFO][getLabel] pos  81 145
[INFO][getLabel] the dffs are 3 5 total 8
[INFO][getLabel] adding label rake
[INFO][getLabel] cog  157 84
[INFO][getLabel] pos  81 145
[INFO][getLabel] the dffs are 76 61 total 137
[INFO][getLabel] cog  168 190
[INFO][getLabel] pos  81 145
[INFO][getLabel] the dffs are 87 45 total 132
[DEBUG][getToolLikeNames] Adding rake 
[DEBUG][getToolLikeNames] res 81 145 
[DEBUG][getToolLikeNames] tempPoints 81 145 
[DEBUG]Adding
[INFO][askPraxicon] tool names: rake 
[INFO][askPraxicon] object names: cheese stick rake bun-top bun-bottom 
[INFO][askPraxicon] sending 
 (available cheese stick bun-top bun-bottom) (query "make a sandwich") (missing) 
```

vtikha, 2015-10-20:
@gsaponaro thanks for your last test and I apologise for the back and forth debugs. I have modified the code and committed it. The changes results in a much more stable extraction of the tool like object labels, and a correct instruction is now sent to the praxicon. I have still left in the debugging info, to see what you are getting from your end. Please try it out when you'll have time and let me know the output.

gsaponaro, 2015-10-20:
Thanks, no problem!

I made a few tests and the propagation is correct now. :)

**Test 1**

[image]

```
>>askPraxicon "make a sandwich"
Response: (((hand grasp cheese) (cheese reach bun-bottom) (hand put cheese) (hand grasp bun-top) (bun-top reach cheese) (hand put bun-top)))
```

```
[INFO][askPraxicon] request is : make a sandwich 
[DEBUG][getToolLikeNames] Number of contours is 5 
[DEBUG][getToolLikeNames] Contour[0] -X[140.291962]  -Y[202.756577] Area: 1950.00 Length: 157.52 
[DEBUG][getToolLikeNames] Contour[1] -X[192.345444]  -Y[147.048859] Area: 1211.00 Length: 131.48 
[DEBUG][getToolLikeNames] Contour[2] -X[264.231476]  -Y[176.912216] Area: 2346.50 Length: 296.47 
[DEBUG][getToolLikeNames] Contour[3] -X[60.179482]  -Y[163.287674] Area: 3030.00 Length: 370.56 
[DEBUG][getToolLikeNames] Contour[4] -X[159.477737]  -Y[93.593155] Area: 696.00 Length: 96.28 
[ERROR][getToolLikeNames] number of tool like objects found 2 
[INFO][getLabel] cog  194 147
[INFO][getLabel] pos  263 172
[INFO][getLabel] the dffs are 69 25 total 94
[INFO][getLabel] cog  264 168
[INFO][getLabel] pos  263 172
[INFO][getLabel] the dffs are 1 4 total 5
[INFO][getLabel] adding label stick
[INFO][getLabel] cog  141 203
[INFO][getLabel] pos  263 172
[INFO][getLabel] the dffs are 122 31 total 153
[INFO][getLabel] cog  159 94
[INFO][getLabel] pos  263 172
[INFO][getLabel] the dffs are 104 78 total 182
[INFO][getLabel] cog  66 161
[INFO][getLabel] pos  263 172
[INFO][getLabel] the dffs are 197 11 total 208
[DEBUG][getToolLikeNames] Adding stick 
[DEBUG][getToolLikeNames] res 65 157 
[DEBUG][getToolLikeNames] tempPoints 263 172 
[INFO][getLabel] cog  194 147
[INFO][getLabel] pos  65 157
[INFO][getLabel] the dffs are 129 10 total 139
[INFO][getLabel] cog  264 168
[INFO][getLabel] pos  65 157
[INFO][getLabel] the dffs are 199 11 total 210
[INFO][getLabel] cog  141 203
[INFO][getLabel] pos  65 157
[INFO][getLabel] the dffs are 76 46 total 122
[INFO][getLabel] cog  159 94
[INFO][getLabel] pos  65 157
[INFO][getLabel] the dffs are 94 63 total 157
[INFO][getLabel] cog  66 161
[INFO][getLabel] pos  65 157
[INFO][getLabel] the dffs are 1 4 total 5
[INFO][getLabel] adding label rake
[DEBUG][getToolLikeNames] Adding rake 
[INFO][askPraxicon] tool names: stick rake 
[INFO][askPraxicon] object names: cheese stick bun-bottom bun-top rake 
[INFO][askPraxicon] sending 
 (available cheese bun-bottom bun-top) (query "make a sandwich") (missing) 
```

**Test 2 with objects closer to each other**

[image]

```
>>askPraxicon "make a sandwich"
[... 108 lines cut ...]
[INFO][getLabel] cog  183 166
[INFO][getLabel] pos  223 145
[INFO][getLabel] the dffs are 40 21 total 61
[INFO][getLabel] cog  128 207
[INFO][getLabel] pos  223 145
[INFO][getLabel] the dffs are 95 62 total 157
[INFO][getLabel] cog  165 111
[INFO][getLabel] pos  223 145
[INFO][getLabel] the dffs are 58 34 total 92
[INFO][getLabel] cog  223 147
[INFO][getLabel] pos  223 145
[INFO][getLabel] the dffs are 0 2 total 2
[INFO][getLabel] adding label rake
[INFO][getLabel] cog  121 111
[INFO][getLabel] pos  223 145
[INFO][getLabel] the dffs are 102 34 total 136
[INFO][getLabel] cog  130 150
[INFO][getLabel] pos  223 145
[INFO][getLabel] the dffs are 93 5 total 98
[DEBUG][getToolLikeNames] Adding rake 
[INFO][askPraxicon] tool names: "" rake 
[INFO][askPraxicon] object names: cheese bun-bottom bun-top rake tomato meat 
[INFO][askPraxicon] sending 
 (available cheese bun-bottom bun-top tomato meat) (query "make a sandwich") (missing) 
```

gsaponaro, 2015-11-13:
Hi @vtikha, any comment about my latest experiments and debug information reported above? My conjecture was that the problem is fixed (however, see my **Sidenote**).

vtikha, 2015-12-14:
@gsaponaro The tool is still recognised due to its characteristics but if the recognition is unstable there is no way to put the correct label. What do you prefer having an empty slot or none at all? In any case the object recognition should be well trained (even if it gets good results) not with just one instance.

gsaponaro, 2015-12-14:
I prefer no slot at all (no empty string), thanks.

closed 2015-12-14 by vtikha