robotology/poeticon #159: PRAXICON reasoner module

gsaponaro, 2015-09-22:
In late July, Panagiotis sent us the link to a new version of the reasoner, solving a bug where instead of providing the naming of the actual object, the reasoner returned the naming of the theoretical object.

Then he asked us:
*The repo you have the poeticon code is open to everyone?
Is it the place for the Praxicon reasoner modules?
I remind you here that the Praxicon reasoner is a netbeans based group of java projects.*

@vtikha, do you think it's practical/useful to commit the PRAXICON reasoner engine here on this repo?

vtikha, 2015-09-22:
@gsaponaro yes, so that Panagiotis can update things easier.

gsaponaro, 2015-09-22:
OK, then allow me to elaborate. We all agree that sharing and synchronizing is good, but this thing is pretty big. :) **How** should we share it? Perhaps something obvious is escaping me...

The last time, Panagiotis packaged a new virtual machine image and uploaded it to BitBucket.

We can continue like that (for external storage [Git Large File Storage](https://git-lfs.github.com/) is probably more easily integrated to GitHub), or we can do something like committing only the actual source code files on the repo, but I would need help for that, as I don't know about the PRAXICON code details.

It's possible that, as we do more tests, we make more frequent requests for changes in the reasoner to Panagiotis. Are we OK with him uploading a new version of the VM every time? (Perhaps we are, bandwidth is cheap :) ) or are we seeking a more "agile" collaboration for this, if so how?

/cc @lorejam @AlexAntn

vtikha, 2015-09-22:
@gsaponaro a new version of the VM? My bad i assumed we were on the same page. I thought it was obvious not to share the virtual machine. Only a few files from the praxicon need to be changed when Panagiotis changes things. This is why i was ok with committing these files. Llet's keep the original format: Panagiotis shares the files as he has always done in the last years with any online sharing method he chooses. At this stage of the project this is not useful.

closed 2015-09-22 by vtikha