robotology/poeticon #164: praxiconInterface crash after receiving sequence of instructions from PRAXICON

gsaponaro, 2015-09-28:
Steps to reproduce:

* on Windows machine: deploy **PraxiconWebService** in NetBeans (we are using the version that Panagiotis shared in his 30 July email); run **praxiconInterface** (the version that is now on this repository)
* on Linux cluster: run object recognition (we are using IOL with Caffe), then launch **activityInterface**
* connect everything
* send query:
```
yarp rpc /activityInterface/rpc:i
>>askPraxicon "make a sandwich"
```

I can see that the query is going through well:
```
[INFO][askPraxicon] request is : make a sandwich 
[INFO][askPraxicon] object names: cheese stick rake bun-top bun-bottom meat tomato 
[INFO][askPraxicon] sending 
 (available cheese stick rake bun-top bun-bottom meat tomato) (query "make a sandwich") (missing) 
```

and **praxiconInterface** actually receives the sequence of actions from PRAXICON... but then crashes like this:

[image]

Notes:
* probably **praxiconInterface**  is missing a check somewhere? (e.g., if a Bottle is formed according to expectations)
* I tried compiling **praxiconInterface** in both Debug and Release, same outcome;
* when I click the "Close program" button in the screenshot, **activityInterface** enters the portion that produces the speech "something went terribly wrong, I cannot make a sandwich".

vtikha, 2015-09-28:
@gsaponaro Now I remember! This is related to the #158 issue. This is why the function was more complex. The issue is that the PRAXICON cannot deal with tools names and is expecting only the "ingredients". In fact it makes sense, it should not care about tools (rake etc...) as they are not part of the sandwich. The planner with the activityInterface should take care of dealing with tools if object is far etc... Now this is why I was removing some objects (tool names) from the list before sending it to the praxicon, hence the missing objects! Now that it is clear for me, I will go back and make sure all tools are removed properly from the query. 
As (see output of terminal):

```
hand - grasp - stick .... stick - reach - cheese ... hand - put - stick 
```
makes absolutely no sense in the sandwich making procedure. 

Two things:

 - There is still obviously a bug in there somewhere as tools are not removing things correctly. 
 - I will also check why it crashed like that, probably size of the Bottle but I am glad it crashed so we could check this tool business.

Great, will fix it asap.

gsaponaro, 2015-09-28:
OK cool, thanks. There was a reason for the complexity. :)

vtikha, 2015-09-29:
@gsaponaro Is the windows machine on? I will need to access it remotely to check this out.

vtikha, 2015-09-30:
@gsaponaro issue #164 fixed and committed.

closed 2015-09-30 by vtikha

gsaponaro, 2015-09-30:
OK, thanks.

gsaponaro, 2015-10-02:
Hi @vtikha, just wanted to report that in a scenario with the two tools (stick and rake) plus several objects, **activityInterface** is correctly "removing" the rake in the ```askPraxicon``` query propagation, but it is not "removing" the stick.

```
INFO][askPraxicon] tool names: rake 
[INFO][askPraxicon] object names: cheese stick rake bun-top bun-bottom meat tomato 
[INFO][askPraxicon] sending 
 (available cheese stick bun-top bun-bottom meat tomato) (query "make a sandwich") (missing) 
```

vtikha, 2015-10-02:
@gsaponaro this should be in #158. have you tried like i mentioned previously? Reduce the number of obj so you have more space on the table and make sure everything is well separated. Like i mentioned i  this has to do with the query of object labels using 2D points (for the tool like objects). I cannot look at this for a few days as i am not available.

gsaponaro, 2015-10-02:
Yes, I forgot to mention that the objects were well separated. Will test some more, check the coordinates and in case open a new issue with screenshots / debug. OK, talk to you later. :)