robotology/poeticon #166: "I made a sandwich" is not spoken anymore

gsaponaro, 2015-10-28:
Today the robot made a meat sandwich, but the spoken words "I made a meat sandwich" were not spoken as they used to.
(During the rest of the plan, speech was working fine, for example "OK I will take the bun-top" etc. was correctly spoken).

Probably, some communication or RPC command is wrong between [PlannerThread::planCompletion()](https://github.com/robotology/poeticon/blob/master/src/pradaPlanner-cpp/planningCycle/src/PlannerThread.cpp#780) and [ActivityInterface::processPradaStatus()](https://github.com/robotology/poeticon/blob/master/src/activityInterface/src/activityInterface.cpp#L471).

The planner printed the following:

```
[DEBUG]Sequence found: put_14_on_16_with_11() grasp_14_with_12() put_14_on_16_with_12() put_14_on_13_with_11() doNothing() 
[DEBUG]put bun-top meat
[INFO]Request execution of action: put bun-top meat
[INFO]Received reply: [ok]
[INFO]Planner state updated!
[INFO]Goal updated
[INFO]PRADA config file reset
[INFO]Plan completed!!
```

/cc @AlexAntn @vtikha

vtikha, 2015-11-02:
@gsaponaro This has not been changed from the activityInterface side. Please make sure that the planner is sending something and have a look at its contents. Then look at the output from the activityIF side.

gsaponaro, 2015-11-05:
Thanks @vtikha.

@AlexAntn: when you can connect one of these days, let's debug this together. Please start by inspecting [PlannerThread::planCompletion()](https://github.com/robotology/poeticon/blob/master/src/pradaPlanner-cpp/planningCycle/src/PlannerThread.cpp#L780).

AlexAntn, 2015-11-06:
Hello @vtikha, @gsaponaro ,

I haven't changed that function (PlannerThread::planCompletion()) ever since it worked. I checked the ports, they should be open, the bottle should be made correctly, and so on.

I don't have access to a Ubuntu-powered computer yet, hopefully next week i will. The only thing i can think of to check this for now is, like @vtikha said, to see if the planner is sending something on that port: /planner/prax_inst:o.

gsaponaro, 2015-11-06:
OK, I'll investigate this next week.

gsaponaro, 2015-11-13:
@AlexAntn, I reproduced the problem with a simple plan, where the objects on the table were: bun-bottom, bun-top, meat (no tools).

What happens is that the planner does not form the Bottle for activityInterface correctly. The formed Bottle contained just "OK", instead of the expected "OK"+names of ingredients.

I would appreciate your help in debugging the function ```planCompletion()```, because I see that it uses several variables (some of them shared with other functions, i.e., ```objects_used```, ```object_IDs```) and nested cycles, so it is not easy for me to debug.

Some more information for you:

```
>>printObjects 
Response: ((11 left) (12 right) (13 bun-top) (14 meat) (15 bun-bottom))
```

```
[DEBUG]Sequence found: put_13_on_14_with_11() grasp_13_with_12() put_13_on_14_with_12() put_13_on_14_with_11() drop_13_with_11() 
[DEBUG]put bun-top meat
[INFO]Request execution of action: put bun-top meat

[INFO]Received reply: [ok]
[INFO]Planner state updated!
[INFO]Goal updated
[INFO]PRADA config file reset
[INFO]Plan completed!!
[DEBUG]string sent to activityInterface prada:i port: OK  <-- wrong

[INFO]PRADA config file reset
```

Feel free to commit more debug prints in the code (do a git pull first). I can do more tests today and in the coming days.

After we solve this simple case (no tools), I will also test the scenario with tools. But first things first. :)

AlexAntn, 2015-11-16:
@gsaponaro I made a commit changing some stuff on the loadUsedObjs function (which is responsible for creating the list of used objects).

Whenever you run another full plan, send me a print of the debug (like the last one you sent), thanks! :)

gsaponaro, 2015-11-17:
Hi @AlexAntn, I ran the same setup and plan, including the latest changes of https://github.com/robotology/poeticon/commit/6bf8f71bfd8305ff760ddc9554fe7aa593ef69b6 and https://github.com/robotology/poeticon/commit/7855161a6db749dc71b862a52d36b97afdf6bfe3

The problem is still present and the debug print is very similar, here it is: https://gist.github.com/gsaponaro/900e9010866990d54f87

```
>>printObjects 
Response: ((11 left) (12 right) (13 bun-top) (14 meat) (15 bun-bottom))
```

I noticed that your 2 recent commits didn't include any new ```yDebug()``` instructions, you could add those now so we can further debug this. Also, a minor thing: in your next git commit message please also write "```see #166```" so it will be automatically cross-linked on the GitHub issue page.

gsaponaro, 2015-11-19:
Hi @AlexAntn, I tested the changes of https://github.com/AlexAntn/poeticon/commit/d702115f404f9e85a3c00665a26a1ff2ae1abae8

Good news: simple plans without tools now generate the proper speech string:
```
string sent to activityInterface prada:i port: OK meat bun-bottom bun-top
```

I also tried to test a plan with tools (several times) but something but the coordinates associated with askForTool were not valid for some reason... will take a closer look tomorrow and test again.

Closing issue for now.

closed 2015-11-19 by gsaponaro