robotology/poeticon #167: Continuous perception probabilities

gsaponaro, 2015-11-06:
Hi @lorejam and @vtikha, in the past we discussed about using a measure of probability/uncertainty associated to some of the perception information.

For me the ideal thing would be if **activityInterface** exposed these scores via RPC, then the World State would use them with filtering/reasoning for PRADA consumption.

Would it still be possible? Or is there no time left?

We could start with the **object reachability** information, which right now causes the biggest annoyances during tests: at the moment an object A is "100% reachable" with B when A is at coordinate X=x cm, but it becomes "unreachable" (0%) when it is at X=x+1 cm. It would be great if **activityInterface** also exposed a continuous score, probably this would be a function of karmaMotor's virtual draw score.

The **activityInterface** RPC function in question is: ```Bottle reachableWith(1:string objName)```.

Current response from RPC (discrete) | Proposed response from RPC (continuous)
------------ | -------------
returning list of labels usable to reach objName | returning labels and associated scores, in the form ```((obj1 prob1) (obj2 prob2) …)``` or ```((obj1 obj2) (prob1 prob2))```

Note: in our previous discussion about this topic (emails "[POETICON] - Planner and WorldState - List of modifications" from 15 April onwards) I thought that the most urgent symbol to benefit from probability measures would be the object labels, but thanks to Caffe that is no longer the case :) therefore I suggested object reachability.

gsaponaro, 2015-11-13:
Hi @vtikha and @lorejam, any opinion on this issue/request?

vtikha, 2015-12-14:
@gsaponaro The fact that the `reachableWith` behaviour is fixed was decided last time I was in Lisbon, therefore there are no probabilities. We can fine tune the system to work as best as possible with respect to the workspace of the robot but without any probabilities. Now the system is tuned so that an object located further than 55 cm from the robot is `not` reachable, which is kinematically sound.

closed 2016-01-14 by gsaponaro