robotology/poeticon #174: Compatibility with OpenCV 3

gsaponaro, 2015-11-30:
@AlexAntn reports some errors while compiling the project with OpenCV 3. For the moment I have specified in the main CMakeLists.txt that we rely on version 2.4 (https://github.com/robotology/poeticon/commit/1a679469182fa6ddf93bcdb5f5cbdd413226cde0).

If we decide to make the code compatible with 3.0, here is the error, related to the macro ```max(a, b)``` in the active segmentation modules:

```
In file included from /usr/local/src/robot/poeticon/src/activeSegmentation/src/library/include/iCub/segmentationLayer.h:37:0,
                 from /usr/local/src/robot/poeticon/src/activeSegmentation/src/library/include/iCub/activeSeg.h:45,
                 from /usr/local/src/robot/poeticon/src/activeSegmentation/src/library/src/activeSeg.cpp:19:
/usr/local/include/opencv2/core/affine.hpp: In member function ‘cv::Affine3<T>::Vec3 cv::Affine3<T>::rvec() const’:
/usr/local/src/robot/poeticon/src/activeSegmentation/src/library/include/iCub/savgol.h:56:19: error: expected unqualified-id before ‘(’ token
 #define max(a, b) ((a)>(b)?(a):(b))
                   ^
/usr/local/src/robot/poeticon/src/activeSegmentation/src/library/include/iCub/savgol.h:56:19: error: expected unqualified-id before ‘(’ token
 #define max(a, b) ((a)>(b)?(a):(b))
                   ^
/usr/local/src/robot/poeticon/src/activeSegmentation/src/library/include/iCub/savgol.h:56:19: error: expected unqualified-id before ‘(’ token
 #define max(a, b) ((a)>(b)?(a):(b))
                   ^
/usr/local/src/robot/poeticon/src/activeSegmentation/src/library/src/activeSeg.cpp: In member function ‘ActiveSeg::Error ActiveSeg::getTemplateFromSeg(const IplImage*, IplImage*, IplImage*&, SegInfo&)’:
/usr/local/src/robot/poeticon/src/activeSegmentation/src/library/src/activeSeg.cpp:143:22: error: conversion from ‘IplImage* {aka _IplImage*}’ to non-scalar type ‘cv::Mat’ requested
     cv::Mat imgMat = seg;
```

/cc @vtikha

vtikha, 2015-11-30:
We do not support OpenCV3 yet in the whole yarp + iCub projects. Switching to OpenCV3 is planned but will not be done in the very near future.

gsaponaro, 2015-11-30:
OK, thanks.

closed 2015-11-30 by gsaponaro