robotology/poeticon #175: activityInterface pullableWith query gets stuck

gsaponaro, 2016-02-11:
Hi @vtikha, I ran into this situation with the latest version of all the software repositories:

* segmentation is good and stable (relying on ```lbpExtract```):
[image]

* **activityInterface** responds correctly to these queries:
```
>>inHand rake
Response: none
>>reachableWith rake
Response: (bun-top left)
```

* however, it gets stuck on the ```pullableWith``` query (for any object), for example:
```
>>pullableWith rake
```

I restarted everything and reconnected just in case. The problem is consistent.

I also tried adding some debug prints in the **activityInterface** to locate the problem, and indeed it gets stuck in the first line of the function ```pullableWith(const string &objName)```, here:
```
    Bottle list = getToolLikeNames();
```

vtikha, 2016-02-15:
Quick question as I have a small doubt. Did you connect correctly the new port ```lbpExtract``` to activity interface? As it get stuck while waiting for an image.. the port should be ```/lbpExtract/extractedlbp:o``` or even ```/lbpExtract/segmentedlbp:o```. Let me know before taking this any further.

gsaponaro, 2016-02-15:
Hi @vtikha, I think that might be the problem, [our XML](https://github.com/robotology/poeticon/blob/master/app/scripts/poeticon%2B%2B_perception_mode.xml.template) does not have such a connection, sorry. Will confirm in one hour.

gsaponaro, 2016-02-15:
I confirm that we are missing two connections: to ```/activityInterface/blobImg:i``` and ```/activityInterface/blobs:i```. Can you please tell me what goes to each of those? (which lbpExtract ports)

vtikha, 2016-02-15:
```/lbpExtract/extractedlbp:o``` goes to ```/activityInterface/blobImg:i```
```/lbpExtract/blobs:o``` goes to ```/activityInterface/blobs:i```

gsaponaro, 2016-02-15:
Problem solved, thanks! :)
Still, I would suggest to add some activityInterface internal checks (getInputCount()>0 etc.) to make the rpc server return [fail] when there are missing connections

closed 2016-02-15 by vtikha