robotology/poeticon #181: activeParticleTrack - crash upon "pause" RPC command

gsaponaro, 2016-02-26:
Hi @vtikha, we have a crash with ```activeParticleTrack``` when it receives the RPC command ```pause(1:i32 id)``` of an object that is not seen (because our robot head joints sometimes are faulting, thus the robot is looking in weird directions). Anyway, we reproduced it a few times:

```
[DEBUG][TRACKERManager::pauseTracker] attempting to pause tracker id 
 14 
[DEBUG][TRACKERManager::pauseTracker] done pausing tracker id 
 14 
[DEBUG][TRACKERManager::pauseTracker] attempting to pause tracker id 
 15 
[DEBUG][TRACKERManager::pauseTracker] done pausing tracker id 
 15 
[DEBUG][TRACKERManager::pauseTracker] attempting to pause tracker id 
 13 
Segmentation fault
```

gsaponaro, 2016-02-26:
One thing I observed is that ```activeParticleTrack``` crashes when the user sends a non-existing ID (I know, this shouldn't happen in the first place, I'm checking the world state side :) ). Anyway, here are the steps to reproduce this problem:

launch activeParticleTrack
(no need to connect anything)
in rpc, pause 49
segmentation fault

vtikha, 2016-02-26:
@gsaponaro you should use the getIDs rpc command to get a list of valid ids.

gsaponaro, 2016-02-26:
Thanks, I'm doing it already here:
https://github.com/robotology/poeticon/blob/master/src/worldStateManager/src/WorldStateMgrThread.cpp#L334-L357

Debugging the world state side with some debug prints now.

vtikha, 2016-02-26:
Then if it returns an empty bottle how is it possible that you send IDs that do not exist?

gsaponaro, 2016-02-26:
I think I identified the problem (for an ID that was existing). Example after a clean restart:

World state initialization (the IDs from 13 onwards were given by the tracker):
```
[INFO]short-term memory:
[INFO]id=11 name=left ...
[INFO]id=12 name=right ...
[INFO]id=13 name=bun-top ...
[INFO]id=14 name=rake ...
[INFO]id=15 name=cheese ...
[INFO]id=16 name=bun-bottom ...
```

For some reason activeParticleTracker stops the tracker 15 (not pausing, really stopping):
```
[DEBUG][TRACKERManager::onRead] will now delete TRACKER 
[DEBUG][TRACKERManager::stopTracker] attempting to stop tracker id  15 
[DEBUG][ParticleThread::onStop] start unlocking  
[DEBUG][ParticleThread::onStop] unlocking done  
[ERROR]received a NULL image, skipping frame 
[DEBUG][ParticleThread::threadRelease] start releasing 
[DEBUG][ParticleThread::threadRelease] releasing done 
[DEBUG][TRACKERManager::stopTracker] stopping tracker id  15 done 
[DEBUG][TRACKERManager::onRead] will now delete TRACKER 
```

Then, the next time that activityInterface sends "pause cheese" to worldState, worldState sends the corresponding "pause 15" to activeParticleTrack, but that track does not exist anymore and it crashes:

World state (corresponding code [here](https://github.com/robotology/poeticon/blob/master/src/worldStateManager/src/WorldStateMgrThread.cpp#L2083-L2092)):
```
yarp: Removing input from /activeParticleTrack/target:o to /wsm/target:i
yarp: Removing output from /wsm/tracker:rpc to /activeParticleTrack/rpc:i
[DEBUG]pauseTrack sending query to activeParticleTracker: pause 15 
[WARNING]pauseTrack obtained invalid response:  
```

Tracker:
```
[DEBUG][TRACKERManager::pauseTracker] attempting to pause tracker id 
 15 
Segmentation fault
```

gsaponaro, 2016-02-26:
One thing that is not clear to me is: why is the tracker doing "[TRACKERManager::onRead] will now delete TRACKER"? I.e. what is the logic behind it (i.e. what happened to the track to trigger a delete)?

I haven't seen this crash happen before, and I've been using activeParticleTrack a lot. Perhaps it could be due to our network problems and/or the recent frequency change in the messages sent my activityInterface to IOL? (activityInterface decides what tracks to pause, then it informs worldState, which does the label2id conversion, when worldState sends the "pause" commands to the tracker.)

Anyway, it will be interesting to see if this happens at IIT too.

vtikha, 2016-03-02:
The behaviour here was fine, There have been no changes in the messages sent by activityInterface to IOL...plus there have been no changes whatsoever to the particleTracker since my last visit at IST. This seems to me that the iDs are not well dealt from your side. Anyway, I will check the state this morning here as I have the robot again.

gsaponaro, 2016-03-02:
Thanks for the tests and the extra checks on your side.

I will add checks on my side (quick solution) and in a few days also remove the dependency on activeParticleTrack altogether (better solution): I discussed with @lorejam and it's possible, it's the sensible thing to do. worldState can rely on the tracking exposed by IOL/activityInterface through activity's get2D & getLabel rpc commands.

At the moment our robot has both thumbs broken... soon more tests on our part.

Closing issue.

closed 2016-03-02 by gsaponaro

vtikha, 2016-03-02:
Good, I am now replying ```[fail]``` when incoherent IDs are sent...Therefore, check your system for why this is happening on your side, you must have something not synced. Ok for the code but, please do not make drastic code changes 3 weeks from the review meeting.

gsaponaro, 2016-03-02:
OK, sure. I will double-check thoroughly the whole system before making that kind of change.

One curiosity that I still have: when activeParticleTrack prints *[TRACKERManager::onRead] will now delete TRACKER*, does it mean that it is killing the track of an object, and then possibly it will start tracking it again in the future (but with a different ID)? So far, worldState is sending "pause" and "resume" commands to activeParticleTrack, but never stop or "untrack" commands (perhaps other modules or the internal logic of the tracker are doing that). As a result I was assuming that, during an experiment, if we start tracking the bread with ID=15, that ID still stay constant for that object (throughout the duration of the experiment). Basically I was expecting pause and resume, never untrack & re-track (with a different ID).

vtikha, 2016-03-02:
in fact, ```pause``` and ```resume``` do NOT ```untrack``` and ```stop``` there are no other modules or internal logic that would do such a thing (it makes no sense @gsaponaro ). If you set correctly ```countFrom``` then the IDs are perfectly fine, nothing gets deleted, i checked it all morning. The ```delete tracker``` only happens if you want to ```clone``` a tracker, when separation occurs etc... as this is not what we are looking for, I removed the call to that function as it seems you were doing something to get in there for some reason...Of all the tests I did this morning I never had that happening.

gsaponaro, 2016-03-03:
Hi @vtikha, I implemented the checks on my side, tested in simulation with some camera data dumps (our robot's under repair for a few more hours), I also did some doing nasty things to the tracker for testing (e.g., if I manually kill a track via RPC -> now worldState notices it and prints a warning).

For the record, I found out the reason why this issue happened in the first place: a few weeks ago I changed the world state initialization routine by also calling activeParticleTracker's ```reset``` function in it. I hadn't tested it very well. It turns out that that function was setting some boolean "flag" to false, which executed the clone&delete that weren't supposed to happen.

This looks ok now. Mental note: if I have more problems with activeParticleTracker's ```reset`` in the future, I can instead do a step-by-step ```untrack``` for all the tracked IDs.

Thanks again for the help here.