robotology/poeticon #183: CMake YarpInstallationHelpers and "+" characters in filename

gsaponaro, 2016-03-10:
Hi @drdanz, it seems that [this yarp commit](https://github.com/robotology/yarp/commit/1590423d89f498bbf3f692875373ce0eea777f20) broke the POETICON++ project build, because we have some XML files in [our app/scripts directory](https://github.com/robotology/poeticon/tree/master/app/scripts) that contain the "+" character in the filename, making a regular expression not happy:

```
RegularExpression::compile(): Nested *?+.
RegularExpression::compile(): Error in compile.
CMake Error at /home/gsaponaro/NOBACKUP/yarp/conf/YarpInstallationHelpers.cmake:339 (string):
  string sub-command REGEX, mode REPLACE failed to compile regex
  "^/home/gsaponaro/NOBACKUP/poeticon_upstream/app/scripts/poeticon++_demo.xml.template;/home/gsaponaro/NOBACKUP/poeticon_upstream/app/scripts/segmentation.xml.template;/home/gsaponaro/NOBACKUP/poeticon_upstream/app/scripts/affordancesExploration.xml.template;/home/gsaponaro/NOBACKUP/poeticon_upstream/app/scripts/dummy_poeticon++.xml.template;".
Call Stack (most recent call first):
  app/CMakeLists.txt:26 (yarp_install)
```

Is it possible to fix it on the ```YarpInstallationHelpers.cmake``` side? Or do you suggest that we remove the "+" character from our filenames instead?

drdanz, 2016-03-10:
@gsaponaro The `REGEX` problem is now fixed, but it still fails on my pc with this error.

```
CMake Error at /opt/iit/share/yarp/cmake/YarpInstallationHelpers.cmake:379 (install):
  install PROGRAMS given no DESTINATION!
Call Stack (most recent call first):
  app/CMakeLists.txt:28 (yarp_install)
```

This happens because the `prada` variable is empty, and it is not something that I can fix... calling the CMake command `install()` with an empty list of files is not correct, therefore the same is valid for `yarp_install`

gsaponaro, 2016-03-10:
Thanks! Will fix the other problem.

closed 2016-03-11 by gsaponaro

drdanz, 2016-03-11:
@gsaponaro Sorry, I was wrong... the `install()` command does accept an empty list of files! I just fixed this in `yarp_install()`

gsaponaro, 2016-03-11:
Gotcha! Cheers.