robotology/poeticon #185: Resetting OPC entries using/forcing an ID in the database

gsaponaro, 2016-03-16:
At the moment, when a plan is completed, the reset command of ```worldStateManager``` forces the user to manually restart ```objectsPropertiesCollector --name wsopc --context poeticon --db dbhands.ini --no-save-db --async-bc``` then reconnect all yarp connections. This is cumbersome.

The world state must reset smoothly from one experiment to the other, at the same time making sure that the planner is happy with some entities having the required/hardcoded IDs.

gsaponaro, 2016-03-16:
Idea: do not actually delete the entries in the database (the "strict ID" entries are actually just [these ones relative to robot hands](https://github.com/robotology/poeticon/blob/master/app/conf/dbhands.ini)), instead just re-set their fields/properties. Instead, objects on the table can change ID from one experiment to the other. For example, in the first experiment we can have (11 left) (12 right) (13 bread)... Then in the second experiment it is fine for the planner to have (11 left) (12 right) (20 bread)...

This should work. Trying it now. If if fails, we'll go for the OPC add-with-forced-id idea.

gsaponaro, 2016-03-17:
Partial good news on this issue: resetting OPC the way we want it is done, no need to implement add-with-forced-id.

The less good news: worldState's reset routine almost works (it does between when switching between many different experiment setups with different number of objects in front of the robot) but not always. There is a pending issue while resetting the ```activeParticleTracker``` part which I did not yet solve. Will investigate more asap.

vtikha, 2016-03-17:
:+1: good news

commit a08f7af1 (2016-03-17): [world state]: reset now also resets tracker correctly

Fixes #185, however see also #186
closed 2016-03-17 by gsaponaro