robotology/poeticon #186: activeParticleTrack - reset when some tracks are paused

gsaponaro, 2016-03-17:
Hi @vtikha, I found a small problem in ```activeParticleTrack```. For the purpose of our demo, I already solved this on my side with a workaround (see below) :sweat_smile:  but in my opinion it would be good to address this anyway.

Problem: the "reset" rpc function of ```activeParticleTrack``` works fine when all tracks are active, but:
1. if one or more tracks are paused when a "reset" command is issued, the "getPausedID" list is not emptied - that ID remains present in "getPausedID" even after the reset, although it is deleted correctly from "getIDs";
2. if we are in the above problematic situation, then a client asks "getPausedID" and obtains for example (5) and he tries to "resume 5", the tracker segfaults (recently you added a check to pause/resume only valid IDs, that works, but the issue here seems to be that this ID is in a limbo between valid and deleted);
3. (minor thing: in the Thrift documentation for this module, I suggest to specify that "getIDs" returns all the IDs, both active and paused - at the moment the documetation talks about the IDs being tracked which is a bit ambiguous, the user might be led to think that it only refers to active, not-paused tracks).

My workaround in the world state routine that resets ```activeParticleTrack``` now looks like this:
* first resume all paused tracks (to be precise, all paused tracks that are "valid" i.e. that are both in "getPausedIDs" and in "getIDs");
* then send "reset" command.

vtikha, 2016-03-18:
Hi @gsaponaro 
thanks for notifying. 
1. reset now also clears the pausedIDs list. 
2. this is no longer an issue due to 1.
3. thrift has been modified

closed 2016-03-18 by vtikha

gsaponaro, 2016-03-18:
Cool, thanks.

commit 7ff35e5c (2016-03-18): [activeParticleTrack] This commit fixed #186. Reset now clears pausedIDs and thrift has been modified