robotology/poeticon #187: error compiling poeticon tree

vtikha, 2016-03-21:
I have the following error compiling the poeticon tree after a recent update:
```
Linking CXX executable ../../bin/dummyWorldStateManager
CMakeFiles/dummyWorldStateManager.dir/src/DummyWorldStateMgrMain.cpp.o: In function `main':
DummyWorldStateMgrMain.cpp:(.text.startup+0xd2): undefined reference to `yarp::os::ResourceFinder::configure(int, char**)'
collect2: error: ld returned 1 exit status
make[2]: *** [bin/dummyWorldStateManager] Error 1
```

gsaponaro, 2016-03-21:
Thanks, I reproduced the problem, it seems to be related to [this yarp change](https://github.com/robotology/yarp/commit/169285ca68247e6d0affcd82ffc1a0f13ef808de) and is probably affecting other modules.

vtikha, 2016-03-21:
fixed. Sorry my bad, compiling on a machine without an updated yarp + icub... remember to remove cache if you need to do it on your end.

closed 2016-03-21 by vtikha

gsaponaro, 2016-03-21:
Ah, good, thanks (indeed I had to clean yarp cache and re-build everything to get it fixed).