robotology/poeticon #195: Scenario 4 - semantic changes in the world

gsaponaro, 2016-03-23:
Hi @vtikha @lorejam @AlexAntn, in the scenario when an object completely disappears from the scene (for example if a person removes it from the table - the object is not hidden in the stack), I noticed that, at activityInterface level, that object disappears from the list "getNames" as expected, but the "reachableWith" list for the disappeared object is kept in memory, for example it continues to be "(Stick left right)" even after the object is removed - probably this was a feature requested to satisfy the planner in the past, remembering some properties of objects after they were in the robot hands, @AlexAntn correct me if I'm wrong.

Anyway, to satisfy the planner in scenario 4, the world state must set the reachableWith and the pullableWith symbols of the removed object to an empty list "()".

My question is: should this change be done at activityInterface level (conceptually more correct, but not sure if @vtikha has time), or at world state level (i.e., overriding the not-null list with a null list, which is a bit of a hack, but it's easy)?

If we decide for the second option, I already have [this computation](https://github.com/robotology/poeticon/blob/master/src/worldStateManager/src/WorldStateMgrThread.cpp#L943-L957) that distinguishes between geometric changes (in those cases I don't do anything special, I propagate the reachable etc. symbols that activityIF gives me, which makes scenario 3 work) and semantic changes (not yet working because of this issue). I'm not using those variables yet, for now I'm always propagating whatever activityIF sends as "reachableWith" list.

vtikha, 2016-03-23:
Hi @gsaponaro, well spotted. It is better to do this change in the activityInterface for consistency. I will make the change first thing tomorrow morning. We can keep the second option as a quick solution for a plan b if we notice that things are still not quite correct.

vtikha, 2016-03-26:
:+1: now fixed

gsaponaro, 2016-03-26:
Cool!
@AlexAntn: I don't remember if you need something similar for pullableWith as well, or just reachableWith becoming () is enough for the planner?

AlexAntn, 2016-03-26:
reachableWith is enough for the planner to detect that the object is not there.

closed 2016-04-28 by gsaponaro